slacky1965/tuya_thermostat_zrd #44: Нет соединения после обновления

gelmeone, 2024-12-25:
Прошел процедуру обновления:

`[2024-12-25 23:53:26] info: 	z2m: Finished update of 'heat_hall_flour'
[2024-12-25 23:53:26] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/heat_hall_flour', payload '{"child_lock":"UNLOCK","current_heating_setpoint":18,"heat":"OFF","linkquality":98,"local_temperature":25.5,"local_temperature_calibration":-3,"max_temperature":45,"min_temperature_limit":1,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":20,"weekdays_p2_hour":11,"weekdays_p2_minute":24,"weekdays_p2_temperature":15.5,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"idle","sensor":"AL","system_mode":"heat","update":{"installed_version":268840961,"latest_version":268840961,"state":"idle"},"update_available":null}'
[2024-12-25 23:53:49] info: 	z2m: Device 'heat_hall_flour' was updated from 'undefined' to 'undefined'
[2024-12-25 23:53:49] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":null,"id":"heat_hall_flour","to":null},"status":"ok","transaction":"p388z-2"}'
[2024-12-25 23:54:23] info: 	z2m: Checking if update available for 'heat_hall_flour'
[2024-12-25 23:55:00] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-12-25 23:55:00] info: 	z2m: Disconnecting from MQTT server`

Далее, перезагрузил z2m и на выходе ничего не изменилось. Удалил устройство и на этом всё, z2m термостат не видит

slacky1965, 2024-12-25:
Можно вопрос. Вы что хоть обновляли, какой термостат?

gelmeone, 2024-12-25:
Извиняюсь, забыл написать.
Термостат Moes, в z2m определялся как TZE204_aoclfnxz

slacky1965, 2024-12-26:
Термостат с такой сигнатурой был обновлен несколькими пользователями без каких-либо проблем. 

Перегрузите z2m. При выключенном приборе (но подключенном к 220), зажмите стрелку внизи держите до тех пор, пока не заморгает значок сети. Если моментально не началось подключение, подождите немного. Если ничего не происходит продожительное время, возможно вам не повезло. Свяжитесь со мной в [телеге](https://t.me/slacky1965), я объясню, как его оживить. Но придется припаять три провода.

gelmeone, 2024-12-26:
Вечером постараюсь конечно проверить. Если не получится, тогда уже через телегу свяжусь. Припаять провода, в принципе, не проблема

gelmeone, 2024-12-26:
Есть прогресс. Устройство обнаружилось, но добавилось как неподдерживаемое:

`9

[0xa4c1385a9151e479](http://192.168.28.100:8123/api/hassio_ingress/DC-mt5LU4uC8P4UVlpUbwoWfXcy1s8Ei1d0IXkBQ8M4/#/device/0xa4c1385a9151e479)
0xa4c1385a9151e479
(0xAA97)
Не поддерживается	[Tuya_Thermostat_r01](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)`

closed 2024-12-26 by gelmeone

slacky1965, 2024-12-26:
У Вас прям чудеса какие-то :))

Но я рад, что все хорошо закончилось.

gelmeone, 2024-12-26:
Да, не то слово:)