slacky1965/tuya_thermostat_zrd #48: Дальнейшие обновления прошивки

Nikolay-Ch, 2024-12-26:
А как теперь обновлять прошивки? Можно ли сделать механизм, чтобы Z2M сам понимал, что можно обновить прошивку? Сейчас Z2M показывает, что обновлений прошивки нет.

slacky1965, 2024-12-26:
Можно, но мне лень :))

А так, кидаете новый файл в `z2m/image` и прописываете его в `local_ota_index.js` Если не хотите всегда при этом перегружать z2m, придумайте название файла и переименовывайте всегда в него, например `tuya_thermostat_castom.zigbee` и это же имя прописать `local_ota_index.js`.

Название образа для кастомной прошивки ОТА начинается с 6565.

Возможно, я почитаю, как добавлять прошивки для обновления в сам z2m, но не сегодня :))

Nikolay-Ch, 2025-01-07:
Я про другое сейчас. Он в ОТА не появляется. Надо в конвертере tuya_thermostat.js подправить код экспорта девайса... Так как сами девайсы рапортуют, что их обновлять нельзя, надо прописать явно, что их можно обновлять.
Сейчас стоит так в этом файле (строка 322): `ota: ota.zigbeeOTA`
А надо изменить строку на: 'ota: true'

Ну и для остальных моделей... Тогда они появятся в меню ОТА Z2M. Версия у меня последняя Z2M (уже исправил все конвертеры). Прошивка: 6

Nikolay-Ch, 2025-01-07:
Поправка... force надо убрать, конечно

Nikolay-Ch, 2025-01-07:
Все... Я вовремя force не убрал, оно повторно решило обновлять все... Сделал переконфигурирование и девайс пришел в норму.
В общем, работает нормально, если без force, то что я написал [выше](https://github.com/slacky1965/tuya_thermostat_zrd/issues/48#issuecomment-2575640858)

slacky1965, 2025-01-07:
Николай, очень похвально, что Вы сами во всем разобрались :))

Но, конвертор изменен еще два дня назад, как только вышла версия z2m 2.0.0.

`local_ota_index.json` не нужно менять. Точнее, изменить его, как Вы рекоммендуетет можно, но это повлечет за собой неудобство при очередном обновлении - вам придется каждый раз редактировать это файл и перегружать z2m.

Если Вы пропишате один раз в этом файле

```{
    "url": "images/tuya_thermostat_zrd.zigbee",
}```

То потом Вам нужно просто переименовать файл `6565-0391-XXXXXXXX-tuya_thermostat_zrd.zigbee` в `tuya_thermostat_zrd.zigbee`, потоложить его в `images` и больше ничего править и перегружать не понадобится.

Удачи.

P.S. Как только вышла версия z2m 2.0.0 я все проверил. Я залил в теромстат оригинальную прошивку, затем обновил его на 1.0.07, а потом уцед 1.0.07 обновил до 1.0.08. И добаавлять что-то помимо `url` не обязательно. Все мои прошивки имеют уникальный image type, который не даст обновиться другому устройству.

Nikolay-Ch, 2025-01-07:
@slacky1965, я рад за то, что у Вас последние прошивки обновляются, однако у меня была старая прошивка и она не работала, о чем я и написал...

И, еще одного Вы не учли... Я конвертор брал из последнего Вашего релиза и он тоже не работал... О чем я Выше также написал. По крайней мере мой девайс (_r01) рапортовал Z2M, что он не умеет обновляться. И только изменение в конверторе позволило обновить девайсы. Насчет остальных параметров: manufacturerName и modelId - все может быть, может они и не нужны,  но отдельный блок апдейта прописывать все равно нужно.

slacky1965, 2025-01-07:
Без обид. Просто посмотрите, когда был сделан релиз и когда пришло обновление z2m до верии 2.0.0, и все встанет на свои места. Конечно в релизе не могло быть нового конвертора, он неделю назад опубликован.

closed 2025-01-23 by Nikolay-Ch