slacky1965/tuya_thermostat_zrd #49: Z2M не видит 2 из 4 термостатов после прошивки

arturoprea, 2024-12-28:
Прошил первый термостат, все хорошо подключился обратно, 
прошил второй и его не видит Z2M, 
прошил третий термостат и все хорошо подключился обратно,
и в конец прошил четвертый и после этого Z2M не видет тоже.

Следовал инструкции, после прошивки удалял термостат и перезагружал Z2M.

Встречалась такая проблема? все 4 термостата из одной партии. Модель _TZE204_aoclfnxz

slacky1965, 2024-12-28:
Да, к сожалению, по какой-то, пока не выясненной, причине термостаты не всегда обновляются корректно.

Есть вариант обесточить термостат. Потом подать питание. Если не подключится, то поиграться с кнопкой сопряжения, т.е. жать ее недостаточно долго, а пару секунд, и успевать отпускать, чтобы индикатор сети не успел замограть. Потом уже долго. И так несколько раз. Если все это не поможет, то к сожалению, остается только прошить его по проводам.

Nikolay-Ch, 2024-12-28:
Кстати, я не отключал термостаты из сети в Z2M... Нажал переконфигурировать и все...

slacky1965, 2024-12-28:
Надо проверить.

arturoprea, 2024-12-28:
Отключение от сети помогло! все 4 подключены, теперь прошивая последний 5-й термостат попробую переконфигурировать как предложил Николай не удаляя из Z2M. Как закончится прошивка напишу.

arturoprea, 2024-12-28:
С 5-ым термостатом сбился где-то на 60%, отключил питание, потом перезагрузил Z2M и уже не хотел его прошивать так как говорил что уже стоит последняя версия. нажал на кнопку интервью и потом на реконфигуряцию и уже видет как обновленный термостат и вроде все работает, поменял другой фаил в images чтобы всетаки прошить еще раз от выдает таймоут.

slacky1965, 2024-12-28:
Какую версию заливали?

arturoprea, 2024-12-28:
[image]

arturoprea, 2024-12-28:
после этого скриншота минут через 5 слетел

slacky1965, 2024-12-28:
Работает же. Забейте. Будет новая версия, обновитесь ...

slacky1965, 2024-12-28:
Главное ничего лишнего не обновлять, а то были уже случаи :))

arturoprea, 2024-12-28:
Спасибо!, а то была такая нагрузка от спама что начинал крутить вентилятор от распбери на малых оборотах)

closed 2024-12-29 by arturoprea