slacky1965/tuya_thermostat_zrd #56: Проблемы с гистерезисом...

Nikolay-Ch, 2025-01-01:
Заметил одну особенность, что при подходе к верхней границе температуры, термостат отключается, а сразу после небольшого уменьшения температуры - включается обратно, и по новой, что приводит к быстрому расходованию ресурса реле. На штатной прошивке этого не замечалось.
Может где-то вкралась ошибка в прошивку или гистерезис как-то неправильно по-умолчанию устанавливается?

devbis, 2025-01-01:
@Nikolay-Ch  Какая у вас модель термостата? Небольшое изменение можно узнать поподробнее? Это похоже на значение гистерезиса делённое на 10? Типа 2º -> 0.2º

slacky1965, 2025-01-01:
Я бы очень просил, чтобы прежде, чем что спросить или сообщить о проблеме Вы бы указывали сигнатуру Вашего термостата. Это хотя бы даст возможность понять, о каком куске кода идет речь.

Если бы с гистерезисом была какая-то проблема, он бы отображался неверно. Проверить можно так - настроить его из меню самого прибора и посмотреть, что отобразится в z2m (некоторые термостаты присылают изменения не сразу).

И еще. Программа на модуле zigbee, которую я переписал, не может управлять включением и выключением нагрузки. Это делает сам термостат по каким-то своим правилам. На которые мы просто можем косвенно влиять настройками, не более.

Nikolay-Ch, 2025-01-01:
Проанализировал логи ХА в итоге, получается, что мой термостат (_TZE204_aoclfnxz) воспринимает гистерезис, как 0,5 градуса до границы сверху и снизу. Поэтому, как только температура превышала этот порог (Tмакс - 0,5) то нагрев отключался. Но, как только происходило отключение нагрева, он тут же (секунд через 20-30) включался обратно, так как нижняя граница работала аналогично - при достижении температуры Tмакс - Tгистерезиз + 0,5, нагрев включался...
Я бы программировал гистерезис по-другому, но ладно, возможно это происходит из-за огрубления показаний датчика. Если поведение нагрева не менялось в прошивке, то задачу можно закрывать...

PS
Сейчас поставил гистерезис 2 градуса, вроде термостат начал работать более предсказуемо. Отключает нагрев так же, а включает немного погодя, так как есть время на охлаждение.

closed 2025-01-01 by Nikolay-Ch