slacky1965/tuya_thermostat_zrd #69: Считывание расписаний

kirill-k2, 2025-01-18:
А есть ли возможность считывать расписание?

В оригинальной прошивке было неудобное представление этих расписаний, но их можно было все вместе посмотрить - что облегчало изменения навроде "а вот вечером в воскресенье хочу исправить на пару градусов выше". Сейчас это можно сделать только реконфигурируя условное воскресенье целиком.

slacky1965, 2025-01-18:
z2m умеет запрашивать расписание и термостат его отдает. Но по какой-то причине в z2m не доделали отображение этого расписания.

slacky1965, 2025-01-19:
Я тут включил debug log и получил интерресные данные. Когда расписание приходит после запроса, то z2m это отслеживает и даже предупреждает, что нет конвертора. Наверно можно сделать вывод в exposes. Нужно подумать. Я не сильный мастер в конверторах, если полулчится ...

```
[2025-01-19 12:16:33] debug:   z2m: Received Zigbee message from '0xa4c1381139683757', type 'commandSetWeeklySchedule', cluster 'hvacThermostat', data '{"dayofweek":64,"mode":1,"numoftrans":4,"transitions":[{"heatSetpoint":15,"transitionTime":0},{"heatSetpoint":15,"transitionTime":0},{"heatSetpoint":15,"transitionTime":0},{"heatSetpoint":15,"transitionTime":0}]}' from endpoint 1 with groupID 0
[2025-01-19 12:16:33] debug:   z2m: No converter available for 'Tuya_Thermostat_r05' with cluster 'hvacThermostat' and type 'commandSetWeeklySchedule' and data '{"dayofweek":64,"mode":1,"numoftrans":4,"transitions":[{"heatSetpoint":15,"transitionTime":0},{"heatSetpoint":15,"transitionTime":0},{"heatSetpoint":15,"transitionTime":0},{"heatSetpoint":15,"transitionTime":0}]}'
```

slacky1965, 2025-02-02:
Сделано.

kirill-k2, 2025-02-17:
Спасибо!

closed 2025-02-17 by kirill-k2