slacky1965/tuya_thermostat_zrd #77: No update available for

Andrew55529, 2025-01-22:
Есть термостат c manufacturerName: '_TZE200_aoclfnxz'.
Хочу его попробовать обновить. (Есть запасной) 
Поправил файл конвертера и он определился:
[image]
Но при нажатии проверить обновления, получаю сообщение о том что их нет:

[image]
Вот log:
```
[2025-01-22 20:57:01] info:     z2m: Checking if update available for '0x04cd15fffe26f204'
[2025-01-22 20:57:01] debug:    zhc:ota: [0x04cd15fffe26f204 | TS0601] Checking if an update is available
[2025-01-22 20:57:01] debug:    zhc:ota: [0x04cd15fffe26f204 | TS0601] Using endpoint '1'
[2025-01-22 20:57:01] debug:    z2m: Received Zigbee message from '0x04cd15fffe26f204', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":72,"imageType":5634,"manufacturerCode":4098}' from endpoint 1 with groupID 0
[2025-01-22 20:57:01] debug:    zhc:ota: [0x04cd15fffe26f204 | TS0601] Got request '{"fieldControl":0,"manufacturerCode":4098,"imageType":5634,"fileVersion":72}'
[2025-01-22 20:57:01] debug:    zhc:ota: [0x04cd15fffe26f204 | TS0601] Checking latest image availability, current: {"fieldControl":0,"manufacturerCode":4098,"imageType":5634,"fileVersion":72}
[2025-01-22 20:57:01] debug:    zhc:ota: [0x04cd15fffe26f204 | TS0601] Getting image metadata...
[2025-01-22 20:57:01] debug:    zhc:ota: Downloaded main index
[2025-01-22 20:57:01] debug:    zhc:ota: Loading override index '/opt/zigbee2mqtt/data/local_ota_index.json'
[2025-01-22 20:57:01] debug:    zhc:ota: Getting local firmware file '/opt/zigbee2mqtt/data/images/1141-d3a3-1111114b-tuya_thermostat_zrd.zigbee'
[2025-01-22 20:57:01] debug:    zhc:ota: [0x04cd15fffe26f204 | TS0601] No latest image currently available, current: {"fieldControl":0,"manufacturerCode":4098,"imageType":5634,"fileVersion":72}'
```
В файле local_ota_index.json тоже менял модель. Потом убирал вообще. Добавлял номер версии убирая force. Сейчас выглядит так:
```
[
    {
        "url": "images/1141-d3a3-1111114b-tuya_thermostat_zrd.zigbee",
        "fileVersion": 74
    }
]
```

devbis, 2025-01-23:
@Andrew55529  Адрес термостата 0x04cd15fffe26f204 говорит, что это не Telink (возможно EFR32), а значит прошивка не подойдёт. 
Телинки начинаются на `0xa4c138`. В термостатах  от туи в основном стоят телинки, но бывают и отклонения, как у вас

Andrew55529, 2025-01-23:
@devbis попробую разобрать и глянуть что внутри.

slacky1965, 2025-01-23:
> [@devbis](https://github.com/devbis) попробую разобрать и глянуть что внутри.

Можно не разбирать. Там будет вот такой модуль.

[image]

Andrew55529, 2025-01-23:
@slacky1965 
Печально. Спасибо за ответ.

closed 2025-02-02 by slacky1965