slacky1965/tuya_thermostat_zrd #79: После пропадания питания возврат к предыдущему состоянию

Nikolay-Ch, 2025-01-23:
А можно сделать так, чтобы после пропадания питания можно было бы вернуться к предыдущему состоянию термостата (вкл/выкл)?
Или, хотя бы была настройка, с какого состояния стартовать...

slacky1965, 2025-01-23:
Так вроде это сам термостат делает без всяких настроек ...

Nikolay-Ch, 2025-01-24:
> Так вроде это сам термостат делает без всяких настроек ...
Выключается при включении? Да, я об этом написал выше...
Но, по возможности, хотелось бы сохранения предыдущего состояния, как у Вайфайных версий...

slacky1965, 2025-01-24:
> > Так вроде это сам термостат делает без всяких настроек ...
> 
> Выключается при включении? Да, я об этом написал выше... Но, по возможности, хотелось бы сохранения предыдущего состояния, как у Вайфайных версий...

Нет, не выключается. Был включен, сняли питание. Подали обратно. Он включился, как был перед пропаданием питания. Проверил только что на `_TZE204_u9bfwha0`