slacky1965/tuya_thermostat_zrd #83: Invalid_image - update to 1.0.11

TischenkoArseny, 2025-01-29:
Здравствуйте. Имеется два термостата Tuya_Thermostat_r01, изначально прошитые на версию 1.0.9. При попытке обновление до 1.0.11 один обновился без проблем, второй такой же - z2m: Update of '0xa4c1380cb61c1c3d' failed (Update failed with reason: INVALID_IMAGE). 
Версия z2m - 2.0.2, термостаты одинаковые.

slacky1965, 2025-02-02:
Если Вы опишите свои действия по обновлению более подробней, возможно найдется и ответ ...

TischenkoArseny, 2025-02-02:
1. Скачиваю свежие файлы прошивки([6565-0391-10123001-tuya_thermostat_zrd.zigbee) + конвертор.
2. Переименовываю файл прошивки, аналогично тому названию, которое вбито в local_ota файле.
3. Копирую прошивку + новый конвертор в соответствующие директории.
4. Перезапускаю z2m.
5. Перехожу в раздел ota в z2m.
6. Проверяю обновление прошивки для двух одинаковых термостатов.
7. Жму обновить прошивку - на одном термостате прошивка начинается и успешно проходит, на втором выдает ошибку (Update failed with reason: INVALID_IMAGE).

slacky1965, 2025-02-02:
Чудеса ...

Попробуйте изменить local_ota_index.json на вот такое

```
[
    {
        "url": "images/tuya_thermostat_zrd.zigbee"
    }
]
```

Перегрузить z2m только не забудьте после изменения ...

TischenkoArseny, 2025-02-02:
Так и делал, на втором шаге. Потом пробовал взять предыдущую прошивку (1.0.10) и менять уже не названия файла прошивки, а файл local_ota. Перезагружал после каждого действия, но толку не дало.
Также пробовал передобавлять в z2m термостат, не помогло.

slacky1965, 2025-02-02:
Тогда включаем в логе debug и смотрим, какой он хочет Image Type

[image]

TischenkoArseny, 2025-02-02:
Вроде бы всё совпадает, imagetype - 913.

Строка 197: [2025-02-02 19:02:32] [34mdebug[39m: 	z2m: Received Zigbee message from 'MOES_TS2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":269037569,"imageType":913,"manufacturerCode":25957}' from endpoint 1 with groupID 0
	Строка 198: [2025-02-02 19:02:32] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/MOES_TS2', payload '{"abs_max_heat_setpoint_limit":45,"abs_min_heat_setpoint_limit":5,"brightness":6,"brightness_day":6,"brightness_night":6,"child_lock":"Unlock","deadzone_temperature":3,"eco_mode":"Off","eco_mode_temperature":20,"frost_protect":0,"frost_protect_on_off":"Off","heat_protect":45,"keypad_lockout":"unlock","last_seen":"2025-02-02T16:02:32.257Z","linkquality":18,"local_temperature":25,"local_temperature_calibration":0,"max_heat_setpoint_limit":30,"minSetpointDeadBand":2,"min_heat_setpoint_limit":5,"min_setpoint_dead_band":0.03,"occupied_heating_setpoint":27,"outdoor_temperature":24,"programming_operation_mode":"setpoint","running_state":"heat","schedule_monday":null,"schedule_saturday":null,"schedule_sunday":null,"sensor":"Outer (OU)","system_mode":"heat","update":{"installed_version":269037569,"latest_version":269496321,"state":"available"}}'
	Строка 199: [2025-02-02 19:02:32] [34mdebug[39m: 	z2m: Device 'MOES_TS2' requested OTA
	Строка 214: [2025-02-02 19:02:32] [34mdebug[39m: 	z2m: Responded to OTA request of 'MOES_TS2' with 'NO_IMAGE_AVAILABLE'
	Строка 442: [2025-02-02 19:02:55] [34mdebug[39m: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"MOES_TS2","transaction":"6fkpx-1"}'
	Строка 443: [2025-02-02 19:02:55] [32minfo[39m: 	z2m: Updating 'MOES_TS2' to latest firmware
	Строка 469: [2025-02-02 19:02:55] [34mdebug[39m: 	z2m: Received Zigbee message from 'MOES_TS2', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20250108","swBuildId":"v1.0.09"}' from endpoint 1 with groupID 0
	Строка 470: [2025-02-02 19:02:55] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/MOES_TS2', payload '{"abs_max_heat_setpoint_limit":45,"abs_min_heat_setpoint_limit":5,"brightness":6,"brightness_day":6,"brightness_night":6,"child_lock":"Unlock","deadzone_temperature":3,"eco_mode":"Off","eco_mode_temperature":20,"frost_protect":0,"frost_protect_on_off":"Off","heat_protect":45,"keypad_lockout":"unlock","last_seen":"2025-02-02T16:02:55.326Z","linkquality":90,"local_temperature":25,"local_temperature_calibration":0,"max_heat_setpoint_limit":30,"minSetpointDeadBand":2,"min_heat_setpoint_limit":5,"min_setpoint_dead_band":0.03,"occupied_heating_setpoint":27,"outdoor_temperature":24,"programming_operation_mode":"setpoint","running_state":"heat","schedule_monday":null,"schedule_saturday":null,"schedule_sunday":null,"sensor":"Outer (OU)","system_mode":"heat","update":{"installed_version":269037569,"latest_version":269496321,"state":"available"}}'
	Строка 493: [2025-02-02 19:02:55] [34mdebug[39m: 	z2m: Received Zigbee message from 'MOES_TS2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":269037569,"imageType":913,"manufacturerCode":25957}' from endpoint 1 with groupID 0
	Строка 494: [2025-02-02 19:02:55] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/MOES_TS2', payload '{"abs_max_heat_setpoint_limit":45,"abs_min_heat_setpoint_limit":5,"brightness":6,"brightness_day":6,"brightness_night":6,"child_lock":"Unlock","deadzone_temperature":3,"eco_mode":"Off","eco_mode_temperature":20,"frost_protect":0,"frost_protect_on_off":"Off","heat_protect":45,"keypad_lockout":"unlock","last_seen":"2025-02-02T16:02:55.525Z","linkquality":90,"local_temperature":25,"local_temperature_calibration":0,"max_heat_setpoint_limit":30,"minSetpointDeadBand":2,"min_heat_setpoint_limit":5,"min_setpoint_dead_band":0.03,"occupied_heating_setpoint":27,"outdoor_temperature":24,"programming_operation_mode":"setpoint","running_state":"heat","schedule_monday":null,"schedule_saturday":null,"schedule_sunday":null,"sensor":"Outer (OU)","system_mode":"heat","update":{"installed_version":269037569,"latest_version":269496321,"state":"available"}}'
	Строка 526: [2025-02-02 19:02:56] [34mdebug[39m: 	z2m: Received Zigbee message from 'MOES_TS2', type 'commandImageBlockRequest', cluster 'genOta', data '{"fieldControl":2,"fileOffset":0,"fileVersion":269627393,"imageType":913,"manufacturerCode":25957,"maximumDataSize":48}' from endpoint 1 with groupID 0
	Строка 527: [2025-02-02 19:02:56] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/MOES_TS2', payload '{"abs_max_heat_setpoint_limit":45,"abs_min_heat_setpoint_limit":5,"brightness":6,"brightness_day":6,"brightness_night":6,"child_lock":"Unlock","deadzone_temperature":3,"eco_mode":"Off","eco_mode_temperature":20,"frost_protect":0,"frost_protect_on_off":"Off","heat_protect":45,"keypad_lockout":"unlock","last_seen":"2025-02-02T16:02:56.072Z","linkquality":93,"local_temperature":25,"local_temperature_calibration":0,"max_heat_setpoint_limit":30,"minSetpointDeadBand":2,"min_heat_setpoint_limit":5,"min_setpoint_dead_band":0.03,"occupied_heating_setpoint":27,"outdoor_temperature":24,"programming_operation_mode":"setpoint","running_state":"heat","schedule_monday":null,"schedule_saturday":null,"schedule_sunday":null,"sensor":"Outer (OU)","system_mode":"heat","update":{"installed_version":269037569,"latest_version":269496321,"state":"available"}}'
[... cut ...] "update":{"installed_version":269037569,"latest_version":269496321,"state":"available"}}'
	Строка 604: [2025-02-02 19:02:56] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'MOES_TS2' failed (Update failed with reason: INVALID_IMAGE)","status":"error","transaction":"6fkpx-1"}'
	Строка 605: [2025-02-02 19:02:56] [31merror[39m: 	z2m: Update of 'MOES_TS2' failed (Update failed with reason: INVALID_IMAGE)

slacky1965, 2025-02-02:
Вы уверены, что это именно тот лог? Нужен не из Web-интерфейса, а из Аддона (или как там у Вас это установлено).

[image]

Справа вверху ...

TischenkoArseny, 2025-02-02:
Да, именно оттуда. Потом в notepad++ поиск всех строк с названием термостата MOES_TS2 в моем случае. И дальше уже итог скинул. Могу весь кусок лога залить

TischenkoArseny, 2025-02-02:
Ещё раз попробовал.

[image]

slacky1965, 2025-02-02:
У нас логи разные. Я не вижу части строк у себя, какие есть в Вашем. А судя по последней строчке, термостат ничего не ответил, ошибка по таймауту ...

Четсно, я не знаю, почему так ...

slacky1965, 2025-02-02:
Ну есть еще вариант прошить его проводами, но для этого нужно кое-что приобрести, чтобы сделать программатор ...

TischenkoArseny, 2025-02-02:
Давайте попробую прошить. Имеется usb-to-uart на базе CP2102, он подойдет?

slacky1965, 2025-02-02:
Не, это не поможет. Самый надежный способ купить [вот такую плату](https://aliexpress.ru/item/1005002904091138.html?sku_id=12000022701450115)  (именно kit). Или [вот такую](https://aliexpress.ru/item/1005002767348600.html?sku_id=12000022087479755). И сделать из нее программатор.

TischenkoArseny, 2025-02-02:
Может у Вас ссылочки на озон или wildberries есть? В беларусь с али доставка достаточно долгая) Но если нет, то значит закажу с али и подожду просто.

slacky1965, 2025-02-02:
Только вот [такая](https://zigbee-shop.ru/catalog/goods/1934)

TischenkoArseny, 2025-02-02:
Хорошо, спасибо. Плату заказал. Ориентировочно в конце месяца будет. Что-то ещё дополнительно необходимо?

slacky1965, 2025-02-02:
Больше ничего. Ну паяльник, проводок один припаять на плате, чтобы прошить ее и потом еще три провода уже на термостате.

Плата придет, свяжитесь со мной в [телеге](https://t.me/slacky1965)

liderman, 2025-05-01:
Здравствуйте, а можно для всех инструкцию приложить по прошивке и какие проводки куда припаивать. У меня тоже 4 термостата окирпичились. Платку заказал.

slacky1965, 2025-05-02:
> Здравствуйте, а можно для всех инструкцию приложить по прошивке и какие проводки куда припаивать. У меня тоже 4 термостата окирпичились. Платку заказал.

Все просто. 
Нужен dev-kit TB-03 или TB-04 от компании Ai-Thinker.
Далее идем вот [сюда](https://github.com/pvvx/TLSRPGM). Читаем, паяем пермычку, заливаем в TB-03(4) прошивку, получаем программатор.

Затем от прогрмматора берем несколько сигналов - +3.3v, GND, SWM и RST. И кидаем их на модуль термостата, все совпадает, только SWM (на фото это D4) от программатора на SWS модуля.

[image]

[image]

Потом очищаем память у модуля. Заливаем bootloader по адресу 0, а прошивку по адресу 0x8000

Все.

TischenkoArseny, 2025-05-03:
Обновление до 1.0.16 прошло штатно на всех термостатах. Спасибо!

closed 2025-05-03 by TischenkoArseny