slacky1965/tuya_thermostat_zrd #86: Ошибка отсутствия image при проверке обновления

Inolant, 2025-02-10:
Прошивался на 1.0.12
По инструкции в папку images положил файл 1141-d3a3-1111114b-tuya_thermostat_zrd.zigbee
Все остальное также по инструкции. После перезагрузки z2m и при нажатии Check for new updates  появляется ошибка, что отсутствует 6565-0391-10123001-tuya_thermostat_zrd.zigbee.
Как я понимаю последний используется только для прошивки уже перепрошитого термостата.

Z2M версии 2.1.1

Что делаю не так? Нужно также класть 6565-0391-10123001-tuya_thermostat_zrd.zigbee в images?

Inolant, 2025-02-13:
Нет никаких идей?

Inolant, 2025-02-14:
Спасибо, хоть и не без танцев с бубном, но термостат обновился.

Обновил на 1.0.12 термостат _TZE204_aoclfnxz производства MOES, для которого были изменения в этой прошивке.

После обновления и перепривязки к HA появилась следующая проблема:

В настройках, как и раньше ставлю регулирование теплого пола по внешнему датчику.

В сущность Термостат попадает local temperature, а не outdoor temperature, как было раньше. 
По факту local temperature завышена и не соответствует действительности, т.к. датчик находится внутри термостата.
Соответственно теплые полы не работают.

Еще момент, отдельного outdoor сенсора в mqtt не появилось. Ожидал, что outdoor сенсор на вашей прошивке появится, как на вашем скриншоте.

[image]
[image]
[image]
[image]

slacky1965, 2025-02-14:
Тут такое дело. Внешней температуры на этой модели никогда не было. То, что Вы приводите в пример из описания, это просто скрин для примера. То, что может этот термостат, описано на [страничке этого термостата](https://github.com/slacky1965/tuya_thermostat_zrd/blob/main/doc/thermostats/tuya_thermostat_r01/README.md).

Дальше, про внешний датчик. Термостат выводит в local_temperature тот датчик, который выбран. Если выбран IN, то там будет температура внутреннего. Если OUT, то внешнего. Если ALL, то там какой-то хитрый алгоритм, он плохо описан, что-то типа внутреннего с контролем внешнего.

И еще, мы никаким образом не можем поменять логику работы термостата. Им управляет MCU, которое мы вообще не трогаем. Из-за того, что Вы обновили прошивку на кастомную, работа термостат не меняется. Поэтому, я сильно сомневаюсь, что до перепрошивки Вы могли видеть внешнюю температуру вместе со внутренней. Этот термостат такого не умеет.

Inolant, 2025-02-17:
В итоге я понял, что после обновления  сбилась калибровка датчика Т пола где-то на 3 градуса. Через настройки подправил, сейчас все работает корректно.

И последний вопрос, из данных термостатов, модели с какой сигнатурой отдельно передают сенсор внешней температуры (датчик Т пола)?

P.S. Спасибо, за ваш труд!

slacky1965, 2025-02-17:
Полчитайте описание конкретных термостатов в самом начале Readme.

[image]

Где будет вот так, как на скрине ниже, там есть двторая температура ...

[image]