slacky1965/tuya_thermostat_zrd #87: Не синхронизируется время, даже после перехода с прошивки v1.0.7 на v1.0.12

arturoprea, 2025-02-10:
Заметил что после пару обновлении Zigbee2mqtt время показывает ошибочно, даже минуты не совпадают.
Перешел с прошивки v1.0.7 на v1.0.12 после, пробовал перезапустить все но все равно показывает ошибочно.
Сейчас 22:38 а на них 19:30
Z2M 2.1.1-1
5 термостатов Moes bht-006
как можно решить проблему?

[image]

slacky1965, 2025-02-15:
Я тут подумал. А он вообще нормально работает? Т.е. если менять параметры, в z2m это видно?

Если нет, то возможно нужно просто обесточить на несколько секунд термостат и все заработает ...

arturoprea, 2025-02-15:
В общем не знаю как но через день само решилось, время начало показывать.

closed 2025-02-15 by arturoprea