slacky1965/tuya_thermostat_zrd #88: Moes bht-006 выключение экрана

dixanl, 2025-02-11:
Термостат Moes bht-006 прошивка  v1.0.12  настроен чтобы выключал экран полностью после 10сек бездействия.
все работает! но если включить на регуляторе защиту от детей то перестаёт выключать экран! но если это сделать через ХА то все отрабатывает. и если экран выключен то при нажатии он не реагирует совсем(не показывает даже значек замка) но иногда включает подсветку правой части экрана где сенсорные кнопки!

dixanl, 2025-02-11:
прошит был сразу на v1.0.12 обновлений не делалось!

slacky1965, 2025-02-11:
> Термостат Moes bht-006

Это ничего не говорит. Какая сигнатура было до обновления?

А Вы уверены, что он до обновления так же не работал?

Объясню, почему я так думаю.  Прошивка меняет только zigbee модуль, который не управляет напрямую термостатом. Вся логика работы зашита в отдельном микроконтроллере, на который мы повлиять не можем.

И, к сожилению, такого термостата у меня нет, проверить не смогу ...

dixanl, 2025-02-12:
я так и не понял где посмотреть какая сигнатура у терморегулятора. он был новый самое интересное что до прошивки он не запоминал значение настройки чтобы выключать экран. и всегда горел. на фото мой термостат 

[image]

slacky1965, 2025-02-15:
Это нужно было поглядеть до прошивки термостата.

[image]

closed 2026-01-01 by slacky1965