slacky1965/tuya_thermostat_zrd #90: Исчез из Z2M и больше не коннектится...

Nikolay-Ch, 2025-02-15:
Внезапно отключился от Зигби сети.
Попытки переспаривания результата не меняют. Иконка спаривания моргает, но в Z2M девайс не залетает.
Остальное - все работает. Экран температуру показывает, релюшкой щелкает.
Как думаете, что можно сделать?

Модель [такая](https://github.com/slacky1965/tuya_thermostat_zrd/blob/main/doc/thermostats/tuya_thermostat_r01/README.md) - черная.

slacky1965, 2025-02-17:
Обесточте его на несколько секунд ...

demushkin, 2025-03-02:
в z2m 2.1.3-1 термостат отваливается в состояние "Не поддерживается" (откат на 2.1.1-1 лечит)
p.s.watermeter v1 тоже отваливается.

slacky1965, 2025-03-02:
[image]

Nikolay-Ch, 2025-03-15:
> Обесточте его на несколько секунд ...

Ничего не помогает... Функционально - он работает, считает температуру, щелкает релюшками. Но сеть ЗигБи не видит. :(

slacky1965, 2025-03-15:
Функционально он будет работать, даже если модуль Zigbee физически из него удалить. В своих экспериментах, я еще ни разу не встречался с ситуацией, когда я не смог бы заджойнить термостат. Да, иногда с танцами, с бубном, в виде - обесточить прибор. Могу посоветовать припаять два провода и проверить через usb-uart адаптер, что там с термостатом происходит. Но только это нужно делать через ноут, чтобы исключить возможное отсутствие гальванической развязки на самом термостате.

Вот сейчас у меня в руках термостат, который мне передали, правда у него сигнатура другая. Пользователь тоже не смог подключить его к сети. Вот буду проверять, что там ..

slacky1965, 2025-03-15:
У термостата, который мне передали, по какой-то причине умер бутлоадер. Разбираюсь.

slacky1965, 2025-03-15:
Мне выслали вот такой вот термостат - `_TZE204_aoclfnxz` . Во всяком случае сказали, что за последнее время с разницев в месяц были куплены у продавца на Али несколько штук и они все пришли с такой сигнатурой. Приедет, посмотрим, что там и как ...

bazuchan, 2025-03-16:
> Мне выслали вот такой вот термостат - `_TZE204_aoclfnxz`

У меня такие, перешил и начали периодически отваливаться от сети. Переподключение каждый раз - час плясок, соединятся попытки с тридцатой.

Nikolay-Ch, 2025-03-16:
> Мне выслали вот такой вот термостат - `_TZE204_aoclfnxz` . Во всяком случае сказали, что за последнее время с разницей в месяц были куплены у продавца на Али несколько штук и они все пришли с такой сигнатурой. Приедет, посмотрим, что там и как ...

У меня два таких... Один отвалился и больше не джойнится, второй - работает. Если умер бутлоадер, он же не должен загружаться и отвечать на команды (типа долгого нажатия на кнопку вниз для начала процесса пейринга)

slacky1965, 2025-03-16:
> У меня два таких... Один отвалился и больше не джойнится, второй - работает. Если умер бутлоадер, он же не должен загружаться и отвечать на команды (типа долгого нажатия на кнопку вниз для начала [...]

Загружаться в сеть не должен, а на кнопки реагировать должен. Этим управляет не zigbee модуль, а MCU, которое отвечает за всю логику работы термостата и нажатие кнопок в том числе. Если Zigbee отвалилось, например, как у Вас, то термостат просто превращаетстя в "не умный". Но работать будет.

К сожалению так же легко его обратно не оживить. Придется припаять 3 провода и купить небольшую платку (в районе 350-400 рублей), чтобы сделать из нее программатор. Поверьте, это не так сложно ...

slacky1965, 2025-03-16:
> > Мне выслали вот такой вот термостат - `_TZE204_aoclfnxz`
> 
> У меня такие, перешил и начали периодически отваливаться от сети. Переподключение каждый раз - час плясок, соединятся попытки с тридцатой.

Я не хотел Вам навредить. Извините. Но я могу Вас научить, как вернуть все назад ...

bazuchan, 2025-03-17:
> > > Мне выслали вот такой вот термостат - `_TZE204_aoclfnxz`
> > 
> > 
> > У меня такие, перешил и начали периодически отваливаться от сети. Переподключение каждый раз - час плясок, соединятся попытки с тридцатой.
> 
> Я не хотел Вам навредить. Извините. Но я могу Вас научить, как вернуть все назад ...

Лучше расскажите как  подебажить)

slacky1965, 2025-03-17:
> Лучше расскажите как подебажить)

Да нет у TLSR8258 нормального дебага. Есть только printf, как у Ардуино.

[image]

Если подпаяться двумя проводами в PB5 и GND, то можно посмотреть, выводит что-ниубдь он в терминал. На 115200. Только лучше через ноут на батарейках, чтобы не было связи с 220 ...

Nikolay-Ch, 2025-03-17:
> > > Мне выслали вот такой вот термостат - `_TZE204_aoclfnxz`
> > 
> > 
> > У меня такие, перешил и начали периодически отваливаться от сети. Переподключение каждый раз - час плясок, соединятся попытки с тридцатой.
> 
> Я не хотел Вам навредить. Извините. Но я могу Вас научить, как вернуть все назад ...

Ну... лучше скажите, это затронет второй экземпляр? Нужно ли мне срочно обновиться на последнюю версию, или откатить назад?

А с первым (который умер) есть ссыль на платку на Алике? И дальше шаги какие?

kirill-k2, 2025-03-19:
> У термостата, который мне передали, по какой-то причине умер бутлоадер. Разбираюсь.

@slacky1965 для информации, может поможет понять, что происходит.

У меня кирпич уже пару раз образовывался в ОКОЛО следующих случаях:

1. Слабый сигнал, плохой коннект, приходилось раз десять вручную перезапускать в Z2M обновление.
2. Рестарт устройства по питанию ПОСЛЕ начала скачивания обновления, но ДО перезагрузки. 

Последний раз, по факту, была последовательность 1+2. Причем устройство нормально рестартануло с новой прошивкой. Но при повторном принудительном обновлении (точнее, отмены скачивания обновления практически в начале) и перезагрузки по питанию перестало откликаться и подключаться к Z2M.

slacky1965, 2025-03-19:
> [@slacky1965](https://github.com/slacky1965) для информации, может поможет понять, что происходит.
>
> У меня кирпич уже пару раз образовывался в ОКОЛО следующих случаях:
>
> 1. Слабый сигнал, плохой [...]

Спасибо. Я примернго представляю, что происходит. Уже кое-что придумал. Но окончательно жду именно этого термостата, чтобы уже конкретно с ним проверить ...

closed 2026-01-01 by slacky1965