slacky1965/tuya_thermostat_zrd #100: Значок моргает, но к ZigBee не подключается.

eqasxar, 2025-03-14:
Привет, казалось что принял более привычный метод и надежный. Прошить Thermostat MOES _TZE204_aoclfnxz через USB адаптер по этой методике https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version. Прошил V1.0.14 и V1.0.12 на разных скоростях. Термостат как бы живой но сеть не ловит. Индикатор моргает не ZHA , не z2m не обнаруживает. Прочистил кеш браузера, ребут Home Assistent, пере подключил питание термостата. Связи нет.

eqasxar, 2025-03-14:
Спасибо, всё понятно. Окирпичил девайс. Заказим другой.

slacky1965, 2025-03-14:
Ну вот зачем? :))

Заказывать новый термостат не нужно. Закажите лучше TB-03 или TB-04, но только не модуль, а KIT - плату с пинами. Сделайте из него [программатор](https://github.com/pvvx/TLSRPGM) и залейте нормально прошивку ...

Оригинальную прошивку пришлю, если нужно ...

slacky1965, 2025-03-14:
Кстати, если он Вам не нужен, пришлите его мне. На них очень много нареканий, а у меня такого нет :))

Nikolay-Ch, 2025-03-15:
Ну вот у меня такая же ситуация (см. [задачу 90](https://github.com/slacky1965/tuya_thermostat_zrd/issues/90))
Тоже моргает индикатором и к сети не подключается.

slacky1965, 2025-03-15:
> Привет, казалось что принял более привычный метод и надежный. Прошить Thermostat MOES _TZE204_aoclfnxz через USB адаптер по этой методике [...]

Кстати, тут подумал :))

А Вы bootloader прошивали?

eqasxar, 2025-03-16:
Нет, только этот tuya_thermostat_zrd_V1.0.14.bin.

slacky1965, 2025-03-16:
Попробуйте прошить с адреса 0.

eqasxar, 2025-03-20:
> Попробуйте прошить с адреса 0.

а это как?  
Для забавы заказал программатор tb03. Но поскольку с ним еще ни когда не работал, то особо на какой то прогресс не надеюсь.

kirill-k2, 2025-03-27:
> а это как? 

примерно так: 
https://github.com/slacky1965/tuya_thermostat_zrd/blob/a7a91b19bafd39162527f0ae588084823f6e63c9/makefile#L170

обратите внимание на ноль ближе к концу. больше примеров - в том же файле и в документации.