slacky1965/tuya_thermostat_zrd #117: Integration of new thermostat Tervix Tuya " _TZE200_6kijc7nd" and "_TZE204_6kijc7nd"

lionkingQ12, 2025-05-08:
Hello, I have 4 thermostats from  Tervix Tuya  _TZE200_6kijc7nd and  _TZE204_6kijc7nd which spams my network much. How can i add this thermostats as models to this custom firmware? Booth models IEEE starts with 0xa4c13886  how to integrate them to this custom firmware.

alexander-saratovcev, 2025-05-08:
Hi @lionkingQ12 , based on information from the README your thermostats are not supported yet.

lionkingQ12, 2025-05-08:
I understand that there is no support yet, so I am writing about it and saying that it is needed.

lionkingQ12, 2025-05-10:
Tell me what is needed for this.   They look exactly the same.
https://moeshouse.com/products/zigbee-002-thermostat-for-heating-boiler?variant=39802433011793

slacky1965, 2025-05-10:
First, you need to disassemble the thermostat and send a photo of the low-current board.

[image]

lionkingQ12, 2025-05-13:
I understand, I'll take the thermostat, disassemble it, and take a photo.

lionkingQ12, 2025-05-20:
[image]
[image]

slacky1965, 2025-05-28:
I can add a thermostat only if it has a ZT3L module installed.

slacky1965, 2025-05-28:
Great. But again - remote addition of thermostats is not supported.

But you can connect a log analyzer to RX-TX, save the readings and send them to me. Readings are needed for each command.

[image]

[image]

ThxAndBye, 2025-07-13:
I got a Tuya-Gateway in and did some new captures with via the official Tuya app. 
Here is the list from the app for the names of the settinfg:

[image]

[tuya_termostat.zip](https://github.com/user-attachments/files/21203552/tuya_termostat.zip)

Let me know if they are good and I can create missing captures to complete them for this model.

closed 2026-01-01 by slacky1965