slacky1965/tuya_thermostat_zrd #119: MOES TS0601 (_TZE200_aoclfnxz) запрос поддержки (support request)

nezg, 2025-06-01:
Hi,

У меня есть 2 _TZE200_aoclfnxz MOES, вижу что поддерживается _TZE204_aoclfnxz.
По вашему опыту, какая разница между 204 и 200?
Есть ли шанс что прошивка заработает если подменить ОТА наименование?
Готов к экспериментам, програматор есть, но времени заниматься не очень много.

I have 2 _TZE200_aoclfnxz MOES, I see that _TZE204_aoclfnxz is supported.
In your experience, what is the difference between 204 and 200?
Is there a chance that the firmware will work if you change the OTA name?
I'm ready to experiment, I have a programmer, but I don't have much time to study.

slacky1965, 2025-06-02:
Нет, не заработает. `_TZE200_aoclfnxz` это другой модуль `Zigbee`. Прошивка для `Telink` не подойдет к `SiLabs`.

ildmgithub, 2025-09-27:
Тоже имею термостаты TZE200_aoclfnxz. Для них в будущем ожидается прошивка?

slacky1965, 2025-09-27:
Нет.

closed 2025-09-27 by slacky1965