slacky1965/tuya_thermostat_zrd #121: MOES ZHT-002 / TS0601: _TZE204_xalsoe3m

ThxAndBye, 2025-06-11:
I've recently got a _TZE204_xalsoe3m and externally it looks like it's similar or even identical to the _TZE204_aoclfnxz.
IEEE is starting with 0xa4c138
Is there a way to verify that the firmware is working with this new hardware revision?

slacky1965, 2025-06-23:
Each command or setting needs to be in a separate file. Speed uart - 9600

Example

[image]

ThxAndBye, 2025-06-27:
I've captured a bit more traffic from the serial bus. I've toggled the settings in Z2M and then captured the serial.
Can you tell me if they are working for you in this way or should I change strategy (e.g. by changing the settings on the device directly)?

[tuya_termostat_xalsoe3m.zip](https://github.com/user-attachments/files/20953816/tuya_termostat_xalsoe3m.zip)

slacky1965, 2025-06-27:
It is best to check on the original Tuya hub. Z2M may have an incorrect converter.

ThxAndBye, 2025-06-27:
I unfortunately don't own a Tuya Zigbee hub currently. If Z2M isn't ideal and before purchasing even more stuff (and if that's also fine on your end) then I could sponsor you to buy another one from the same AliExpress shop (or buy one and have it shipped to you). Unfortunately thermostats seem to be on the EU sanctions list as dual-use goods, otherwise I could've send you my already prepared unit. 

Otherwise I'd need to buy a [Tuya Zigbee-Hub](https://de.aliexpress.com/item/1005005625247295.html) (like this one?) as the next course of action.

ThxAndBye, 2025-07-13:
I got a Tuya-Gateway in and did some new captures with via the official Tuya app. 
Here is the list from the app for the names of the settinfg:

[image]

[tuya_termostat.zip](https://github.com/user-attachments/files/21203552/tuya_termostat.zip)

Let me know if they are good and I can create missing captures to complete them for this model.

closed 2026-01-01 by slacky1965