slacky1965/tuya_thermostat_zrd #122: Ошибка при обновлении zh:ember:uart:ash received error from adapter with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT

Exitialis, 2025-06-15:
### Контекст

Добрый день. Понимаю, что к самой прошивке здесь не относится и дело в чём-то другом. Однако, я уже потерял всякую веру разобраться в том, что происходит, перечитал кучу статей и провёл миллион экспериментов.

При запуске обновления через OTA процесс стартует и всё идёт, % обновляется. Однако, через некоторое время насыпает кучу ошибок и адаптер отваливается, процесс обновления на этом зависает.

[image]

[image]

Sonoff Zigbee 3.0 USB Dongle Plus V2.

При обычной работе zigbee сети всё при этом работает, но сеть у меня пока небольшая, т.к. я её только строю и купил контроллеры тёплого пола, чтобы были роутерами, но они начали засирать все логи и zigbee2mqtt начал ругаться на это, поэтому и ставлю прошивку.

### Что я пробовал

1. Обновлял прошивку Zigbee dongle через сайт https://darkxst.github.io/silabs-firmware-builder/
2. Обновлял прошивку как вручную на разные версии, так и автоматически.
3. Нашел issue, где чувак [рекоммендует определённую версию прошивки](https://github.com/Koenkk/zigbee2mqtt/issues/23761). Постави, помогло, теперь отваливается не через пару секунд, а через пару десятков секунд, но процесс обновления всё равно не удаётся пока завершить.
4.  Обновление ставлю при отключённых контроллерах теплого пола, работает только один, который и обновляю.
5. Отключил 2.4 Ghz в WiFi роутере
6. Dongle подключён через usb удлинитель в сервер. Сервер HP Microserver Gen 8, на нём стоит виртуалка HAOS
7. Конфиг zigbee2mqtt такой:
`
version: 4
homeassistant:
  enabled: true
  legacy_action_sensor: true
permit_join: false
advanced:
  channel: 11
  network_key:
    - 175
    - 44
    - 244
    - 23
    - 84
    - 176
    - 35
    - 188
    - 223
    - 166
    - 71
    - 147
    - 47
    - 158
    - 8
    - 37
  pan_id: 54515
  last_seen: ISO_8601
  log_level: info
  ext_pan_id:
    - 242
    - 145
    - 122
    - 53
    - 132
    - 87
    - 39
    - 40
mqtt:
  server: mqtt://core-mosquitto:1883
  base_topic: zigbee2mqtt_main
  user: addons
  password: wiekei2beihaequ4mucoomahni8ies5choLaSh2ookaig6phu4thoocahsai6ahc
serial:
  port: >-
    /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_ac7d3ece8039ef119bd159f454516304-if00-port0
  adapter: ember
frontend:
  enabled: true
  port: 8099
ota:
  zigbee_ota_override_index_location: local_ota_index.json
devices:
  '0x54ef441000bd532e':
    friendly_name: Розетка конвектор
  '0xa4c138d6e69d581e':
    friendly_name: Шторы гостиная
  '0xa4c1381685e4c2f6':
    friendly_name: Над столом
  '0xa4c13800c5b061ef':
    friendly_name: Кухня
  '0x54ef441000bd4ec8':
    friendly_name: Розетка камера
  '0x00158d000af43fcf':
    friendly_name: Температура балкон
  '0x00158d000af43f52':
    friendly_name: Температура мастер сан узел
  '0x00158d000af43f27':
    friendly_name: Температура детский сан узел
  '0x00158d008b03e422':
    friendly_name: Температура гостиная
  '0x00158d008b3aab70':
    friendly_name: Датчик протечки кухня под раковиной
  '0x00158d008b3aad5b':
    friendly_name: Датчик протечки кухня под плинтусом
  '0x00158d008b3ab54f':
    friendly_name: Датчик протечки мастер санузел коммуникации
  '0x00158d008b3b0812':
    friendly_name: Датчик протечки мастер санузел унитаз
  '0x00158d008b3aad3d':
    friendly_name: Датчик протечки мастер под раковиной
  '0x00158d008b3b0684':
    friendly_name: Датчик протечки детский санузел коммуникации
  '0x00158d008b3aad43':
    friendly_name: Датчик протечки детский санузел под раковиной
  '0x00158d008b3a9e0d':
    friendly_name: Датчик протечки детский санузел под унитазом
  '0xa4c13885278d760f':
    friendly_name: Теплый пол кухня
    homeassistant: {}
    optimistic: true
    debounce: 5
  '0xa4c138bc25d79666':
    friendly_name: Теплый пол кабинет
    debounce: 4
    homeassistant: {}
    optimistic: true
  '0xa4c1384149cef565':
    friendly_name: Теплый пол лоджия
    debounce: 4
    homeassistant: {}
    optimistic: true
`

Карта сети такая:

[image]

### Мольбы о помощи

Подскажите, что ещё можно попробовать, что я делаю не так? К сожалению нигде не могу найти информацию, что ещё можно попробовать.

Exitialis, 2025-06-15:
Сам девайс _TZE204_aoclfnxz

[image]

Exitialis, 2025-06-15:
От версии прошивки модуля в итоге ничего не зависит, кинул удлинитель USB так, что stick теперь находится в 3х метрах прямой видимости от контроллера тёплого пола, это ничего не даёт также. Пытаюсь отключить сейчас остальные устройства из сети, буду искать виновника. Хотя по логам события шлёт только сам контроллер тёплого пола (каждые 2 секунды по 25 событий)

slacky1965, 2025-06-15:
[image]

https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update5

Но это не точно, но человеку помогло ...

Exitialis, 2025-06-15:
Да, я как раз разобрался, но сюда отписать не успел. Поставил прошивку [отсюда](https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.2-update5/sonoff_zbdonglee_zigbee_ncp_8.0.2.0_115200_sw_flow.gbl) на версию 8.0.2 и это кажется решило проблему. По крайней мере пока так и выглядит.

Спасибо за помощь

closed 2025-06-15 by Exitialis