slacky1965/tuya_thermostat_zrd #123: Need external converter or not?

clinkme4g, 2025-07-19:
There is stated in the description, 
that "As of April 2025 version of zigbee2mqtt no external converter is needed. Support is enabled globally".

Unfortunately, the thermostat with this new firmware doesn't work with z2m version 2.5.1  and zigbee-herdsman-converters version  23.69.0.
I've should use the second external converter from git, which described the thermostat as 
Slacky-DIY | Tuya_Thermostat_r05.

Is that normal, and if not, how I can eliminate this?

slacky1965, 2025-07-20:
I specially connected the same thermostat. Everything works correctly.

[image]

[image]

clinkme4g, 2025-07-26:
What I did:
1)      Remove the current thermostat instance from z2m
2)      Delete all .js-files from z2m folder external_converters/
3)      Power cycle the thermostat
4)      Restart the entire z2m container
5)      Pair again the thermostat 
Now it introduces itself as THERM_SLACKY_DIY_R05 and works ok.
Thanks!

closed 2026-01-01 by slacky1965