slacky1965/tuya_thermostat_zrd #124: New thermostat _TZE204_gops3slb

HD78, 2025-07-19:
I have received earlier few _TZE204_oh8y8pv8 which installed fine as R6 Thermostat.

Now I have received new ones, having id _TZE204_gops3slb. Externally perfectly identical to previous ones so should be the R6 I think.
Is there easy way to force them to work also?

slacky1965, 2025-07-20:
https://github.com/slacky1965/tuya_thermostat_zrd/issues/117#issuecomment-2917492651

HD78, 2025-07-21:
I'll see if I can pull that data from it. Needs plenty of work, including making the log analyzer to tap into it.
Just thinking a quick and dirty method, is there a way to force it to get the r06 firmware? I could risk it with one and try.

HD78, 2025-07-23:
I can give it a try. If it breaks it breaks..

slacky1965, 2025-08-24:
> I tried to flash oh8y8pv8 FW into the gops3slb. I though that it will then act as the old oh8 version but no, it's still identified as the gop. version. BTW the original gops3slb will be supported in the next release of Z2M [Koenkk/zigbee2mqtt#28035](https://github.com/Koenkk/zigbee2mqtt/issues/28035)

The device name is given by the main microcontroller.

slacky1965, 2025-08-24:
https://github.com/slacky1965/tuya_thermostat_zrd/tree/_TZE204_gops3slb/bin

If the update is via OTA, then the second time you need to update from this repository with the `file 6565-0391-10163001-tuya_thermostat_zrd.zigbee`, and not the global one from `zigbee2mqtt`.

Benik3, 2025-08-24:
hmm something screwed, but I'm not sure if it has anything to do without your firmware.
I flashed the tuya_thermostat_zrd_V1.0.16.bin
But I'm not able to get the thermostat into pairing mode (by holding up arrow). The module beeps, but the icon of signal doesn't start to flash.
What is the most weird, that even flashing back the backup of the original firmware didn't help...

slacky1965, 2025-08-24:
Try turning off the power for 10 seconds.

HD78, 2025-08-24:
I flashed mine as OTA so maybe that is the reason mine works and yours doesn't?

slacky1965, 2025-08-24:
> I flashed mine as OTA so maybe that is the reason mine works and yours doesn't?

What is the current version number of your thermostat?

slacky1965, 2025-08-24:
> > What is the current version number of your thermostat?
> 
> It's at v1.0.16. I flashed the [6565-0391-10163001-tuya_thermostat_zrd.zigbee](https://github.com/slacky1965/tuya_thermostat_zrd/blob/_TZE204_gops3slb/bin/6565-0391-10163001-tuya_thermostat_zrd.zigbee) also.

Please show me the "About" tab in z2m.

HD78, 2025-08-24:
[image]

Benik3, 2025-08-24:
I successfully flashed the thermostat from oh8 FW to yours 1.0.16 :)
Everything looks working except 2 settings:
Schedule mode: Z2M report this error
[image]

Max heat setpoint limit: Not sure if it's only communication problem. When I try to set the value using HA, it shows only 80°C or 86°C on the display and another icon light up. After that, I'm not able to change it under 80°C using arrow down on the thermostat, but using arrow up and going over 95°C successfully set the temp to the 15°C and setting looks working fine again.

HD78, 2025-08-25:
Now I tested this to another device I have actually in use at the moment. Used the new 1141-d3a3-1111114b-tuya_thermostat_zrd which showed it right away as R06 version with fw v1.0.00. But I did run into same problem as Benik3, the unit wouldn't pair anymore. Powered it off from circuit breaker for >15mins, then it paired again first try.
Updated to v1.0.16 succesfully and it is in full operational use.

Benik3, 2025-08-25:
I suppose that's normal. I had to also delete and rebind the unit after first OTA flash. I suppose that the new (custom) firmware is not able to reuse the saved data from the original firmware.
For me the rebinding worked fine and the unit bind automatically after power on.

Benik3, 2025-10-17:
It looks like the device could be supported in future version of Z2M:
https://github.com/koenkk/zigbee2mqtt/issues/29062
If it will work, I would recommend to capture the data on the Tx/Rx lines (https://github.com/slacky1965/tuya_thermostat_zrd/issues/117#issuecomment-2917492651) of the ZT3L module and then ask slacky1965 for support.
I also recommend to create separated issue for this device to avoid mess in this issue for gops3slb.

closed 2026-01-01 by slacky1965