slacky1965/tuya_thermostat_zrd #125: Feature request: Brightness control

Tee86, 2025-07-24:
Hi Slacky,

Firstly, great work with all of this! Awesome really...

Do you reckon it'd be possible to add a NumberSelector that would allow user to control brightness of the device? Or at least ENUM with `off`, `normal`, and `night` options. I imagine `night` to be a mode where an active brightness would be the level that is normally as `dim`, and `dim` would be `off` -  so the display would be lit only when you touch the device.

I own 9 models of R01 (Moes BHT002) and its light is very disturbing to some people when they're trying to sleep...

slacky1965, 2025-11-14:
This is impossible for R01 thermostat.

luismipr, 2025-11-14:
I have more than 30 r07
with this model, it would be possible??

Thx in advance!!

slacky1965, 2025-11-18:
> [@slacky1965](https://github.com/slacky1965) do we have any of R01-R09 that I could fully turn off or dim at nighttime? Cause even 'eco-mode' in my R01 is quite bright

https://github.com/slacky1965/tuya_thermostat_zrd/blob/main/doc/thermostats/tuya_thermostat_r03/README.md

slacky1965, 2026-01-02:
> If I understand the OP correctly, their thermostat is lit all night and disturbes them when trying to sleep. I don't have this issue, as my Moes turns off the backlight after 10s.

Many thermostats have settings that are not accessible online, but are easily adjusted through the thermostat menu.